FRC2706/blingServer #33: Categorization of fine azumith control of robot

kajames, 2020-01-13:
Work with the software team to determine a way to quickly and accurately point the robot at a specific target.

kajames, 2020-01-14:
The software folks have put together a test program which maps the full range of a joystick to a -0.5 <-> +0.5 range on the motor power. This appears to be usable for easier fine pointing control on protobot.

This needs to be tested by one of last year's drivers (aka Eric)

kevlam2706, 2020-01-14:
This would be tricky given our current space but it would be good to test this on our Deep Space robot on a carpeted area, to see if it makes the robot driving experience better all around - I.e. maybe this should be a permanent change.

There could still be a button to hold down to let loose on the rotation control should it really be necessary.

kajames, 2020-02-08:
Software has completed its investigations

closed 2020-02-08 by kajames